evildmp/BrachioGraph #11: Wiring diagram

AlStal, 2019-10-21:
A simple wiring diagram would be great, for either the servo wiring harness or the breadboard option (where the documentation states the RPi doesn't have enough 5v sources to drive all 3 servos).
Thank you

willward1, 2019-10-23:
I was a bit confused about this also. I got it to work by
1. connecting the + wires of all 3 servos together, and then into one of the 5v pins of the pi
2. connecting all of the ground wires of all three servos,and then into the ground pin of the pi
3. connecting the data wire of all three servos (usually the orange one) and into the GIPO pins 14,15,and 18 of the pi.  

Quick and dirty diagram:
[image]

AlStal, 2019-10-23:
Perfect, thank you

Level0gamedev, 2019-10-24:
Thank you @willward1 - I came here with the exact same question!

evildmp, 2019-11-05:
I found it useful not to touch the wires of the servos themselves, but to splice and solder some ribbon cable extensions.

willprice, 2019-12-01:
[image]

Here's a fritzing diagram I put together

commit a74a5cfd (2022-02-15): Updated documentation

Improved Get started tutorial; added new images.

Closes #11 - thanks @willprice for the diagram.
closed 2022-02-15 by evildmp