evildmp/BrachioGraph #13: Cannot get it to draw anything resembling a box

BrettHolton, 2019-10-26:
Built the BrachioGraph the other day and the results of the bg.box() function look like this

[image]

How do I best troubleshoot what is wrong? 

One thing I noticed was that the drive_xy() movements don't seem to match the text descriptions (ie increase x 1 cm, increase y 1cm) it seems possibly flipped?
 I get values like (-10,-6) for min/max X and (5,11) for min/max y. All of my servo orientations and wiring match the documentation.

The only difference I can think of is that I am using a RaspberryPi 2Bv1 not a zero.

Anyhow Im pretty excited about this project but not sure how best to troubleshoot it. Im going to start reading through the code to build some understanding.

BrettHolton, 2019-10-26:
OK using the bg2 object from bg.py gives much better results and points me in the correct direction. It might be worth adding a troubleshooting section. I assumed that since I bought the exact servos and made my arms 8cm each that it would be close.

evildmp, 2019-10-27:
I think that your issue is that one or both of your motors are going in the wrong direction.

In https://github.com/evildmp/BrachioGraph/commit/d945017eb4f524a9ca85b7832207f1f777d225ec I fixed an issue where I had forgotten to account for the need to handle this.

You could try initialising the BrachioGraph with `servo_1_degree_ms` and `servo_2_degree_ms`, as described in https://brachiograph.readthedocs.io/en/latest/reference.html#the-brachiograph-class.

evildmp, 2019-10-27:
Also, unless your clothes-peg and the other objects in your photo are exceptionally tiny, I don't think the arms you made really are 8cm long - they look more like 8 inches!

BrettHolton, 2019-10-27:
The original in the picture was 16cm x 16cm, I rebuilt it as an 8cm x 8cm
thinking the large size required too much torque.

evildmp, 2019-10-27:
Try initialising with:
```
bg = BrachioGraph(inner_arm=8, outer_arm=8, bounds=(-6, 4, 6, 12), servo_1_degree_ms = -10, servo_2_degree_ms = 10)
```

evildmp, 2019-10-27:
I have: 

* added a new section to the getting started guide https://brachiograph.readthedocs.io/en/latest/get-started/drive.html#check-the-movement
* corrected the default values in the code

I hope this fixes it for you.

BrettHolton, 2019-10-27:
Thank you for addressing this so fast.

closed 2019-10-27 by BrettHolton