evildmp/BrachioGraph #26: Help Creating a new Instance

jjricks6, 2019-11-04:
Hey, I am new to the Raspberry Pi community, and I thought I'd try this as my first project. I am running into a problem where I am trying to create a new Brachiograph instance according to the instructions. The problem comes when I enter:
`from brachiograph import BrachioGraph`
it returns
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/BrachioGraph/brachiograph.py", line 4, in <module>
    import readchar
ModuleNotFoundError: No module named 'readchar'`.

I assume this has to do with the "readchar" package I downloaded in a previous step, so I tried reinstalling it then creating an instance of brachiograph again, but the same problem occurs. In the virtual environment I installed it using the command:
`pip install readchar`
Must I redirect the readchar path to a location that the BrachioGraph program can find it? Am I doing something wrong?

Again, I am new to Raspberry pi and Python, so any help would be appreciated!

bmordue, 2019-11-04:
Have a look at this proposed update to the docs: https://github.com/evildmp/BrachioGraph/pull/24/files
It might be that `pip3 install readchar` (ie, `pip3` instead of `pip`) will resolve your issue?

jjricks6, 2019-11-05:
That did it! Thank you so much bmordue!

closed 2019-11-05 by jjricks6