evildmp/BrachioGraph #29: Another warning against cheap servos

Findus23, 2019-11-04:
I know that the docs already mention it (https://brachiograph.readthedocs.io/en/latest/explanation/hardware.html#beware-of-fakes), but I thought it can't be that bad, after all I just wanted to try it out.

It turns out that the arms can be moved by more than a centimeter before the gears acturally start turning and as a result any movement smaller than a centimeter is just a wiggly line.

https://video.lw1.at/videos/watch/b6bdccd1-6d72-41a4-bca3-c446ce1b1ce8

So next up I'll try [real TowerPro servos with metal gears](https://www.amazon.de/gp/product/B07MY2SWMN/) and hope that I can avoid someone else buying crappy servos.

willward1, 2019-11-04:
+1 
I bought cheap servos off Amazon, wired it all up, and powered it up to be greeted by that horrid gear servo grinding sound in 2/3 of the servos.  Definitely would advise testing new servos before doing any mods to make the wiring harness.

evildmp, 2019-11-05:
@Findus23 The test pattern in your video isn't actually that bad...

Findus23, 2019-11-20:
I tried again today after replacing the servos with TowerPro metal gear ones.
There is definitely an improvement, but still most movements below around 0.7mm don't do anything. I'm not sure to what extent this is to be expected, but looking at the images by others they seem to include details that appear to be impossible with mine.

Longer lines are far better as once it started moving it seems to be more precise but nevertheless one servo got stuck at times.
https://video.lw1.at/videos/watch/88da95b5-1d11-4d11-b65b-9a4e866d22ff

jkrauska, 2019-12-01:
@Findus23 What are you using for arms there? Some sort of metal toy parts?

Findus23, 2019-12-02:
@jkrauska You are right, they are some kind of metal strips I had lying around from a toy many years ago. They are pretty great as the holes are exactly 1 cm apart and allow to combine everything with screws instead of glue.
https://www.winklerschulbedarf.com/at/i/flachstab-7-loch-metall-per-stk-101003

closed 2020-12-05 by evildmp