evildmp/BrachioGraph #33: Help controlling the arms

jjricks6, 2019-11-05:
I started up the BrachioGraph program exactly as stated in the instructions. The instructions told me that when I set the arm lengths that it will initialize the arms to the default position, but nothing moved. 

The console printed something along the lines of "Connection accepted on socket 5" , "So_KEEPALIVE accepted on socket 5" (for sockets 5 - 8 I think). Not sure what this means, but I ignored it for now.

I tried going through the rest of the commands listed in the docs, and nothing happens with the motors. I thought maybe my wiring was bad, or the motors were too cheap, but I tested them with another program and they work like a charm! So I know its not a wiring or motor problem; I must be doing something wrong with the program. Any suggestions will be much appreciated.

(I am using a raspberry pi 4, NOT a zero)

evildmp, 2019-11-05:
Can you share a picture of your wiring?

jjricks6, 2019-11-05:
Yes. Here they are.

Red orange yellow are power
Black grey white are ground
Blue green purple are the controllers

[image]
[image]

evildmp, 2019-11-05:
Assuming that in your first picture, red is power, then it looks OK.

Can you confirm that the _same wiring_ works with the _same motors_ on the _same machine_, but with some other program?

jjricks6, 2019-11-06:
Yes, I tested them using a program I found at this link: https://www.bluetin.io/servo/micro-servo-9g-raspberry-pi-servo-code/ . It worked perfectly.

Just to make sure I am not doing something I am not supposed to, I am posting my console log. 

`pi@raspberrypi:~ $ sudo pigpiod
pi@raspberrypi:~ $ cd BrachioGraph
pi@raspberrypi:~/BrachioGraph $ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.`
`>>> from brachiograph import BrachioGraph`
`>>> bg = BrachioGraph(inner_arm=7.7, outer_arm=8.4)`
`2019-11-05 18:36:36 pthSocketThread: Connection accepted on socket 5`
`2019-11-05 18:36:36 pthSocketThread: SO_KEEPALIVE enabled on socket 5`

`2019-11-05 18:36:36 pthSocketThread: Connection accepted on socket 6`
`2019-11-05 18:36:36 pthSocketThread: SO_KEEPALIVE enabled on socket 6`

`2019-11-05 18:36:36 pthSocketThread: Connection accepted on socket 7`
`2019-11-05 18:36:36 pthSocketThread: SO_KEEPALIVE enabled on socket 7`

`2019-11-05 18:36:36 pthSocketThread: Connection accepted on socket 8`
`2019-11-05 18:36:36 pthSocketThread: SO_KEEPALIVE enabled on socket 8`

`>>> bg.set_angles(angle_1=-90, angle_2=90)`
`>>> bg.drive_xy()`
`-8.7 8.4`
`-9.7 8.4`
`-10.7 8.4`
`-10.7 8.4`
`-10.7 8.4`
`-10.7 8.4`
`-9.7 8.4`
`-9.7 8.4`
`-8.7 8.4`
`w-8.7 8.4`
`w-8.7 8.4`
`w-8.7 8.4`
`w-8.7 8.4`
`-8.7 8.4`

evildmp, 2019-11-06:
I am sure you are not doing anything wrong.

I'd like to find out if anyone else has issues with a Pi 4.

jjricks6, 2019-11-18:
I found the reason why my motors were not working. The problem was because I was using a RPi 4B for the project, but there is a solution. The pre-installed version of pigpio on the pi 4 is not at its latest release. I had to manually update to the newest version. Now the motors work like a charm!

This post gave me the answer I needed. 
https://www.raspberrypi.org/forums/viewtopic.php?p=1497505

closed 2019-11-18 by jjricks6