evildmp/BrachioGraph #44: Brachiograph software under cygwin?

fpiantedosi, 2019-11-14:
Hi all, I have some issue installing python package under windows7, may I use cygwin in order to install python and python packages needed to the project? 

Problems for me are to find correct packages versions that work with 3.7 or 3.8 python versions. 

Thank you
Francesco

optimismus, 2019-11-26:
I am a beginner here and not sure, that I understand what you want, but you have to use a raspberry PI, not Windows.

Findus23, 2019-11-26:
You don't need cygwin to install python on windows. It should work by just installing the python for windows. But of course to control the servos you need something like a raspberry pi.

fpiantedosi, 2019-11-27:
What I needed to do is to install python under cygwin in order to perform "offline" operations as vectorise or json generation. In any case, even if I was not able to install packages under cygwin, I have all I need under windows 7 with python 3.6. 

Installed pip and all other staff including numpy and openCV.

I'm waiting for my raspberry zero, I decided to buy WH model, and I'll try to assembly the plotter.

On raspberry side I'll install GPIO packages in order to use json file to "feed" brachiograph.

Thank you!
Francesco

evildmp, 2019-11-30:
With the new [virtual mode](https://www.brachiograph.art/how-to/virtual-mode.html), you can do quite a bit of experimenting while running on a machine other than a Raspberry Pi. And of course, using ``linedraw.py`` is much faster when on a higher-powered computer.  

If there are issues installing the software on other operating systems it would be good to know.

fpiantedosi, 2019-12-23:
Unable to install under cygwin, fixed installation on WIndows 7

closed 2019-12-23 by fpiantedosi