evildmp/BrachioGraph #48: defining "bounds" (doc error)

optimismus, 2019-11-26:
Hi,
great project!
The documentation "Start up the BrachioGraph" mentions:
```
Reinitialise your plotter with these values:

bg = BrachioGraph(inner_arm=<inner_arm>, outer_arm=<outer_arm>, bounds=<bounds)
```

It tooks me some time to find out, that "bounds" has to deklared like this:
`bg = BrachioGraph(inner_arm=<inner_arm>, outer_arm=<outer_arm>, bounds=(<bounds>))`

Hope that helps!

optimismus, 2019-11-26:
With help from Daniele, I got it working. I initialized my "bg" like this:
`bg = BrachioGraph(inner_arm=8, outer_arm=8, bounds=(-3,5,3,8), pw_up=2300, pw_down=1800)`
with `bg.box()` it draws a box and with `bg.test_pattern()` it draws four lines. Yeah!
If you want to test drawing a json, this command from docu won't work:
`bg.plot_file("test_file.json")`
because, there is no "test_file.json" at /home/pi/BrachioGraph.
You can use this command: `bg.plot_file("/home/pi/BrachioGraph/test-patterns/landing-test.json")`

Findus23, 2019-11-26:
> "bounds" has to be declared like this:

The instructions are technically correct, but maybe not entirely clear. It's `bounds=<bounds>` with `<bounds>` meaning either a list or a tuple (it doesn't matter in this context).
So either `bounds=[1,2,3,4]` or `bounds=(1,2,3,4)`

optimismus, 2019-11-26:
the says `...bounds=<bounds)` and that means (to me) `... bound=1,2,3,4)` which will not work, you need two more brackets.

commit 5989cfd4 (2019-11-26): Clarified bounds, fixed typo

Fixes #48
closed 2019-11-26 by evildmp