evildmp/BrachioGraph #50: ModuleNotFoundError: No module named 'PIL'

optimismus, 2019-11-30:
This doesn't work:
`pi@raspiB:~/BrachioGraph $ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from linedraw import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/BrachioGraph/linedraw.py", line 10, in <module>
    from PIL import Image, ImageDraw, ImageOps
ModuleNotFoundError: No module named 'PIL'`

https://brachiograph.readthedocs.io/en/latest/how-to/use-linedraw.html

As adviced I installed: `pip3 install opencv-python-headless` before.

Regards
Axel

alex8411, 2020-10-08:
same issue when trying to run the Reachy Unity simulation + python webserver API, any solution ?

MushinMiscellanea, 2021-03-27:
Support for PIL was discontinued in 2011. The project is now being maintained as Pillow.

`pip install Pillow`