evildmp/BrachioGraph #57: Unstable Pen - resonances?

jkrauska, 2019-12-01:
When playing around with drive_xy, I find some locations generate unstable behavior.
https://youtu.be/Dn1bVbstcOw
I also see much more shaky lines in the vertical plane than in the horizontal plane.

I tried isolating the elbow servo to see if it would shake like this alone, but it only seems to happen when assembled.

I'm using SG92 servos which generate more torque than SG90s. Perhaps related?

evildmp, 2019-12-04:
Yes, SG92s are digital motors, and very twitchy. See https://www.brachiograph.art/explanation/hardware.html#analog-vs-digital-servo-motors. They are not recommended. 

Even more powerful digital motors need to be used with care. The problem is that the two motors interfere with each other, and set off the oscillation. Analog motors have a dead band that absorbs this tendency.

evildmp, 2019-12-06:
Perhaps also add your findings at https://github.com/evildmp/BrachioGraph/issues/31.

jkrauska, 2019-12-09:
Replaced with [Miuzei SG90 9G](https://www.amazon.com/gp/product/B072V529YD/) and have had much more stable results.

closed 2019-12-09 by jkrauska