evildmp/BrachioGraph #58: "Remote driver" mode?

skorokithakis, 2019-12-09:
Would it be desirable to add a "remote driver" mode to this, where instead of a Raspberry Pi the servos are connected to an ESP8266 and receive instructions over the network from a computer running the Brachiograph script?

That would also make the machine much cheaper (or much more expensive, depending on your definition, but I assume you need a computer now anyway).

evildmp, 2019-12-14:
I'd be interested in seeing a more detailed proposal for this, definitely. I am not sure what the implications are.

tg44, 2020-02-25:
When I saw this project first, my question was the same; why we need an rpi? It could be used in school environments if the zeros cost and relatively high entry knowledge could be lowered (I mean ssh, command line, etc.).

I didn't read the code yet, but I think the biggest problem is the "realtimeness", and if we want to solve that we need to move the coordinate to movement coordination to the mcu, which would result a rather complicated fw and a gcode like driver protocol...