evildmp/BrachioGraph #68: Exception when importing BrachioGraph

littlezaphod, 2020-01-11:
Hi, forgive me if this a trivial question (python beginner) but when importing the module:

>>> from brachiograph import BrachioGraph
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/BrachioGraph/brachiograph.py", line 608
    raise Exception(f"Cannot reach {hypotenuse}; total arm length is {self.INNER_ARM + self.OUTER_ARM}")
                                                                                                      ^
SyntaxError: invalid syntax
>>>

Findus23, 2020-01-11:
Hi,

This is using f-strings which are a new feature in Python 3.6, so you need to use at least this version.

willward1, 2020-01-11:
having same issue.  does anyone have a best practices for upgrading the python  on a pi zero?  I am in the pi forums and it looks like a lot of people are having issues with upgrading the version

littlezaphod, 2020-01-13:
I upgraded python to 3.6. See google for procedure. You have to essentially pull the python sources and build on the Pi. Takes about 1h on a Pi zero. After upgrading you need to pull in the packages needed by brachiograph. 
This got me going, and I am now plotting stuff!
Great project!

closed 2020-01-13 by littlezaphod

Findus23, 2020-01-14:
@willward1, @littlezaphod 

The current stable Debian version (buster) comes with Python 3.7, so it seems like you are still using an older version of Debian.