evildmp/BrachioGraph #69: Syntax error

kenidowa, 2020-01-28:
I have run into a syntax error when loading brachiograph.py at line 608.  
raise Exception(f"Cannot reach {hypotenuse}; total arm length is {self.INNER_ARM + self.OUTER_ARM}")
The f before the " mark doesn't make sense to me.  Is this correct?  I see it in several print statements lower down in the code.  I am not familiar with this syntax and it was flag as an error when I went to Start up the BrachioGraph
@ from brachiograph import BrachioGraph

Findus23, 2020-01-28:
Hi,

This is the exact same thing as #68.

f-strings were introduced in Python 3.6, so you need to use at least this version for them to work.

evildmp, 2020-12-06:
@kenidowa BrachioGraph uses f-strings, and requires Python 3.6 or later.

closed 2020-12-06 by evildmp