evildmp/BrachioGraph #73: AttributeError: 'str' object has no attribute 'wait'

psytel, 2020-03-10:
Hey there!

I am new to python and raspberry so I managed to install everything and now trying to test a svg image. I get this output:

>>> BrachioGraph.plot_file("africa.svg")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/psytel/BrachioGraphVM/BrachioGraph/brachiograph.py", line 136, in plot_file
    wait = wait or self.wait
AttributeError: 'str' object has no attribute 'wait'

Could someone help me? Thanks a lot :)
cheers

meli94, 2020-04-12:
Hey, I have the same issue. Did you manage to fix it?

Anustart1, 2020-05-16:
I have the same problem, I wish someone could help

scholzco, 2020-05-25:
Are there any updates regarding this issue?

john104433, 2021-01-23:
That's right I think.  You have to plot the africa.json file.