evildmp/BrachioGraph #77: 3D printed arms?

VanushVaswani, 2020-04-09:
I love the Lo-Fi nature of this project. Has there been any effort to use 3d printed parts for added stability?

ringo2k, 2020-04-15:
Hi, i am currently working on 3D printed parts, i will host them on thingiverse when i am finished.
[image]

VanushVaswani, 2020-04-15:
Nice, looking forward to that build.

ringo2k, 2020-04-18:
I uploaded it on thingiverse.. this site did a real downgrade during the last update wow.. 

https://www.thingiverse.com/thing:4295302

Hope it looks better on your side.

VanushVaswani, 2020-04-18:
Yep. Thingiverse unfortunately sucks now.

Nice work.. have you tried any prints?

ringo2k, 2020-04-18:
Yes, i make a video and put in on youtube. 
[image]

Thingiverse is beeing unusable atm.. sad.

VanushVaswani, 2020-04-19:
Left a comment on Thingiverse.

ringo2k, 2020-04-19:
Seen and recommented,

oakey1979, 2020-04-19:
Sorry to hijack this thread, but I can't get mine working. The various commands to move the servos result in really odd things happening where the arms thrash wildly. Any ideas where I can start troubleshooting?

VanushVaswani, 2020-04-19:
Did you mount it exactly how the doc said? Show a pic

oakey1979, 2020-04-19:
[image]

oakey1979, 2020-04-19:
I've gone over the guide and rebuilt and rewired it to double check. You can see from the green lines how it moves when I try to move the arms 5 degrees or draw the box as part of the setup.

VanushVaswani, 2020-04-19:
Your shoulder servo looks 180 degrees from mine.

oakey1979, 2020-04-19:
Good spot - I'll rebuild that part, do you think that's the cause of the weird behavior?

VanushVaswani, 2020-04-19:
Yep

VolkerH, 2020-04-20:
Looks very nice @ringo2k, will give this a try as there is way too much sag with my ice cream popsicle sticks.

However, the last few times I tried to download the files I see this
[image]

Not your issue, I will try again.

Thingiverse really has regressed, it would actually be nice to have some Github extensions for 3D prints (simple with OpenSCAD).

mcsage, 2020-04-20:
@VolkerH Download all files doesn't work but single file download does. https://www.thingiverse.com/thing:4295302/files

ringo2k, 2020-04-20:
Yes, thingiverse having real troubles the last days.. I uploaded everything three times because there were files missing.. I am still worried that the upload failed, so let me know. I could also but the STL files on my fork, to host it on github?

oakey1979, 2020-04-20:
I've mounted the shoulder servo the other way around, but I'm still getting crazy behaviour.
See these videos, one is the behaviour just from connecting power, the other from inputting the arm lengths. This also occurs with each subsequent command I use to adjust the angles or draw the box. I've never got any further than this in this project as I have to eventually cut the power when the elbow tries to do a full 360 degree turn and pushes the lifter servo into the shoulder.
https://youtu.be/1rGhffW9mgI
https://youtu.be/ViuSpmldO2g
Any help that can be offered is very welcome!

ringo2k, 2020-04-21:
@oakey1979 the videos are private, I cannot see them. 

Is your power supply ok? Sometimes servos behave strange because when they drive they need a lot of current, which can result in a voltage drop and then things get weird. How much amps does your power supply handle?

oakey1979, 2020-04-21:
I've updated the privacy on the videos. In the video you'll see that this is powered from my laptop. It's a Surface 3, I don't know the power output. I have tried plugging in a second USB cable from a mains charging hub to the pi at the same time and had similar problems.

ringo2k, 2020-04-21:
Oh ok, this really looks strange. When you remove the sticks from the servos, are they still freaking out?

Normally a USB socket can provide 500mA, USB3 900mA. 

I run my Brachiograph with a 3A power supply, a 2A power supply resulted in a restart of the pi. 

Would be interesting to hear what other people are using for powering their build. I mean my Raspi Model 2 is using 300mA in idle, there is no much margin when powered over USB.

VanushVaswani, 2020-04-21:
Try using a 2.5A supply. Some USB chargers may only have 0.5A

ringo2k, 2020-05-01:
Hey, I just uploaded my video on youtube. 

https://www.youtube.com/watch?v=Cv_uGYh3t70

oakey1979, 2020-05-05:
Hi, I've got hold of a powered USB hub and Anker powercore USB cables to ensure there's enough power but my servos are still freaking out like in the videos. I'm about to chuck the whole thing in the bin. Before I do, are there any other suggestions?