evildmp/BrachioGraph #79: ModuleNotFoundError: No module named 'readchar'

oakey1979, 2020-04-11:
Following the guide, step by step. I've got as far as Create a BrachioGraph instance, when, I get this:
pi@raspberrypi:~ $ sudo pigpiod
pi@raspberrypi:~ $ cd BrachioGraph
pi@raspberrypi:~/BrachioGraph $ python3
Python 3.7.3 (default, Dec 20 2019, 18:57:59)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from brachiograph import BrachioGraph
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/BrachioGraph/brachiograph.py", line 4, in <module>
    import readchar
ModuleNotFoundError: No module named 'readchar'

Findus23, 2020-04-11:
Hi,

As the guide mentions (https://www.brachiograph.art/get-started/install.html#other-python-packages) you need to install `readchar` first.

oakey1979, 2020-04-11:
I'm assuming I did that with the line
pip install tqdm readchar ?
I didn't see any errors at the time, but I'll try again without using the Virtual Environment (pip3) method.

Trampf, 2020-04-11:
try pip3

oakey1979, 2020-04-11:
Thanks all, Using pip3 seems to allow me to progress to the next step.

closed 2020-04-11 by oakey1979