evildmp/BrachioGraph #87: Error When Importing The BrachioGraph Library, and Error When Putting in Arm Lengths

Anustart1, 2020-05-15:
Sorry if this is a strange question (python beginner), but when I'm importing 
`from brachiograph import BrachioGraph`
it says:
`>>> from brachiograph import BrachioGraph
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/BrachioGraph/brachiograph.py", line 4, in <module>
    import readchar
ModuleNotFoundError: No module named 'readchar'`
And this is very confusing, as I don't know what this means, and I can't find anything answering it.

My second question might be fixed if the first question is resolved, I don't know, but when I put in 
`bg = BrachioGraph(inner_arm=<8>, outer_arm=<8>)`
it gives me an 'invalid syntax' error, pointing to the beginning of '<8>'
Please help!

Findus23, 2020-05-15:
Did you follow the steps at https://www.brachiograph.art/get-started/install.html#other-python-packages ? There you need to install `readchar`

Don't put `<>` around the numbers:
```python
 bg = BrachioGraph(inner_arm=8, outer_arm=8)
```

Anustart1, 2020-05-15:
> 
> 
> Did you follow the steps at https://www.brachiograph.art/get-started/install.html#other-python-packages ? There you need to install `readchar`
> 
> Don't put `<>` around the numbers:
> 
> ```python
>  bg = BrachioGraph(inner_arm=8, outer_arm=8)
> ```

Ok, so that worked, but I'm still having issues when testing a box, it says this:
`>>> bg.box()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/BrachioGraph/brachiograph.py", line 406, in box
    self.xy(bounds[0], bounds[1], wait, interpolate)
  File "/home/pi/BrachioGraph/brachiograph.py", line 439, in xy
    (angle_1, angle_2) = self.xy_to_angles(x, y)
  File "/home/pi/BrachioGraph/brachiograph.py", line 610, in xy_to_angles
    hypotenuse_angle = math.asin(x/hypotenuse)
ZeroDivisionError: float division by zero`
Why is it doing this?

john104433, 2021-01-09:
I have the same issue with readchar even though I've installed it.  The project uses Python3 but readchar seems to be associated with Python2.7.

Did you get it working Declan?

Declan0811, 2021-01-09:
> I have the same issue with readchar even though I've installed it.  The project uses Python3 but readchar seems to be associated with Python2.7.
> 
> 
> 
> Did you get it working Declan?

No, sadly. I still dont know what is wrong, so I've just given up on it.

john104433, 2021-01-10:
Hello Declan, that's a shame!  I have written simple programs in Python but I'm no expert!  What I have discovered is that you get syntax errors when running in the wrong version of Python for the program you're trying to execute.

To get readchar and tqdm to run I had to do the following:

sudo apt install python3-pip

sudo pip3 install readchar
sudo pip3 install tqdm

So that solved that problem!  pip3 installs for Python 3 and pip for Python 2.  I discovered that on the Raspberry Pi website.
I'll keep you informed as I go along!

John

john104433, 2021-01-11:
So I've started off with a fresh OS.  Followed all the instructions to the letter and checked that readchar is actually present in:

usr/local/lib/python3.7/dist-packages

As soon as I run 'from brachiograph import BrachioGraph' I get this

>>> from brachiograph import BrachioGraph
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/BrachioGraph/brachiograph.py", line 4, in <module>
    import readchar
ModuleNotFoundError: No module named 'readchar'
>>>

Does anyone know why?

john104433, 2021-01-11:
I think I've found it now.  In home/pi/env/pyenv.cfg

I changed the line include-system-site-packages = false to true

home = /usr/bin
include-system-site-packages = true
version = 3.7.3

Now I've progressed!

john104433, 2021-01-11:
Hello Declan, Hope you're still interested!

All the problems and errors I had were all solved by changing that one line indicated above from false to true.

Now I have it working.  Haven't drawn anything with it yet but the servos are all moving so now I'll calibrate it.

At one point I typed bg = BrachioGraph and got all sorts of errors.

I should have typed bg = BrachioGraph()  and then the errors went.

John

doscas, 2021-04-15:
I have the same issue in a different place (trying to use linedraw).

I've followed all the steps and when doing `opencv-python-headless`, I get:
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
` NameError: name 'opencv' is not defined`

Is there something wrong?