evildmp/BrachioGraph #90: Brachiograph in Arduino?

ekaitz-zarraga, 2020-05-30:
Hi,

I'm thinking about making Brachiograph in other devices, like Arduino, for instance.
As the computer has to be connected to it, it should be "simple" to make a smaller or a not-very-powerful device work.

Anyone tried to achieve something similar?

In order to make this work I'd like to know some details I can't answer by myself at the moment, I hope you can help me with that:

- It surprises me that motors are directly connected to GPIO pins, isn't there any kind of current limitation? I've checked motor's "datasheet" (if we can call it that) and it says they drain quite a lot of current (hundreds of mA-s), much more than the PiZero GPIO pins are supposed to deliver. If in the Zero works fine, would it work if I plug them directly to the pins of the Arduino?
- How about using stepper motors from old DVD or CD readers? I suppose they need some reduction to have much more steps per 180º, am I right?

Thanks!

just-jason, 2020-05-30:
> * It surprises me that motors are directly connected to GPIO pins, isn't there any kind of current limitation? I've checked motor's "datasheet" (if we can call it that) and it says they drain quite [...]
Only the signal wires of the servo are connected to the GPIOs of the Pi. Power is taken from the 5v line.
>
> * How about using stepper motors from old DVD or CD readers? I suppose they need some reduction to have much more steps per 180º, am I right?
>  
Possible, but more wires and complications like homing switches, stepper drivers etc....

ekaitz-zarraga, 2020-05-30:
> Only the signal wires of the servo are connected to the GPIOs of the Pi. Power is taken from the 5v line.

Oh, that makes perfect sense.
In the case of the arduino that can be too much too for the 5V line, but I'll try. Thanks.

> Possible, but more wires and complications like homing switches, stepper drivers etc....

Yes. I was wondering about the possible accuracy increase it could have. Surely it will need some gears for reduction and so on so maybe this is not the best use for that kind of motors. They will probably shine more in classic two axis plotters.

Thanks!
I'll give a try to the Arduino and comment here

goldscott, 2020-10-29:
@ekaitz-zarraga any update on getting this working on arduino? I'd like to do the same.

ekaitz-zarraga, 2020-10-30:
> @ekaitz-zarraga any update on getting this working on arduino? I'd like to do the same.

I didn't, yet.
But it shouldn't be veeery difficult.

closed 2020-12-06 by evildmp