evildmp/BrachioGraph #95: pigpio.error: 'pulsewidth not 0 or 500-2500' error

bob486, 2020-09-17:
When I try the tests for the BrachioGraph for instance 1bg.test_pattern()1 it always says :
```
raceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/BrachioGraph/brachiograph.py", line 332, in test_pattern
    self.xy(bounds[0],   y,     wait, interpolate)
  File "/home/pi/BrachioGraph/brachiograph.py", line 475, in xy
    self.set_angles(angle_1, angle_2)
  File "/home/pi/BrachioGraph/brachiograph.py", line 501, in set_angles
    self.set_pulse_widths(pw_1 + self.active_hysteresis_correction_1, pw_2 + self.active_hysteresis_correction_2)
  File "/home/pi/BrachioGraph/brachiograph.py", line 548, in set_pulse_widths
    self.rpi.set_servo_pulsewidth(14, pw_1)
  File "/home/pi/.local/lib/python3.7/site-packages/pigpio.py", line 1677, in set_servo_pulsewidth
    self.sl, _PI_CMD_SERVO, user_gpio, int(pulsewidth)))
  File "/home/pi/.local/lib/python3.7/site-packages/pigpio.py", line 1009, in _u2i
    raise error(error_text(v))
pigpio.error: 'pulsewidth not 0 or 500-2500'
```

How would I fix this?

spheppner, 2020-09-19:
yes i get the same error. please fix this

E0175137, 2020-10-05:
Same, I got stuck on this error, please advise.

bob486, 2020-10-07:
Yeah.

evildmp, 2020-12-05:
The bounds of the box you are trying to draw the test pattern in are too large, and the pulse-widths required to meet them exceed the allowed values (500-2500mS).

The new default BrachioGraph definition https://github.com/evildmp/BrachioGraph/blob/master/brachiograph.py#L19-L40 and clearer getting-started guide should ensure that you stay within limits.

closed 2020-12-05 by evildmp

bob486, 2020-12-11:
I tried the new updated version and it worked. Thank you very much for the help!