evildmp/BrachioGraph #98: I go to check movement and nothing works.

aduncan1157, 2020-10-31:
Power up the Raspberry Pi. Run:
sudo pigpiod
cd BrachioGraph
python3
And then, using the inner_arm and outer_arm length measurements (in cm) that you noted earlier:
from brachiograph import BrachioGraph
bg = BrachioGraph(inner_arm=<inner_arm>, outer_arm=<outer_arm>)
The system will create a BrachioGraph instance and initialise itself, adjusting the motors so that the pen will be at a nominal:
x = -inner_arm
y = outer_arm
And this will correspond to:
the upper arm at -90 degrees, 1500µS pulse-width
the lower arm at 90 degrees to it, 1500µS pulse-width
the lifting motor in the pen up position, 1700µS pulse width
Check the movement
We must make sure that the arms move in the direction we expect. Run:

bg.set_angles(angle_1=-90, angle_2=90)

Once I change the values nothing moves. I triple checked my wiring and even replaced the wires and servos.

nerijus-st, 2020-11-20:
You have to insert your control wires exactly to these pins:
shoulder motor: GPIO pin 14
elbow motor: GPIO pin 15
lifting motor: GPIO pin 18

evildmp, 2020-12-05:
Try one servo at a time.

closed 2020-12-05 by evildmp