evildmp/BrachioGraph #99: Segmentation fault happens when creating a BrachioGraph instance

nebukadezza, 2021-01-16:
My apologies if I'd overlook something obvious. I've strictly followed the software installation steps, but when I've created a BrachioGraph, I got a segmentation fault:

pi@raspberrypi:~ $ sudo pigpiod
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ cd BrachioGraph/
(env) pi@raspberrypi:~/BrachioGraph $ python
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from brachiograph import BrachioGraph
Segmentation fault
(env) pi@raspberrypi:~/BrachioGraph $

The host is running a Raspbian Buster. I appreciate any suggestion on how to get out here, please.

john104433, 2021-01-20:
Well, I Googled that and all the answers say it's a filesystem corruption fault.  I would reflash the SD card.  I too am working through this project and have got as far as drawing a box.  I've also managed to get a couple of clipart images of Andy Capp and Donald Duck drawn. Now I'm struggling with the calibration!

John

John

nebukadezza, 2021-01-20:
Thank you @john104433 , indeed it was the SD card, I've flashed an another one, and it works that way, it makes nice fuzzy lines on paper.

closed 2021-01-20 by nebukadezza